Terrapin-Rocket-Team/Spaceshot #56: Filament Wind

CooperBrowned, 2025-02-26:


CooperBrowned, 2025-02-26:
Current issues: 
- [ ] Tension Belts
- [ ] Figure out homing
- [ ] Clean exopy off gears or print new ones

spacecraft1013, 2025-04-07:
Info that I got from a dude on the r/rocketry discord server: 

The "multi tow" is essentially using multiple rolls of carbon that stay separate until they combine at the end during the wind, here's a video on that: https://www.youtube.com/watch?v=1A3vaJaNDLY. The advantage of this is that the tow is far less likely to bunch up or twist on itself, and can therefore end with a flat, even layer going down on the wind.

They also ran into issues with the rollers fraying, and decided to completely forego the rollers and use static aluminum rods on them, based on this paper: https://core.ac.uk/download/pdf/9837741.pdf (I'll read this tomorrow). Also suggested using epon 862 and curing agent w for their resin since the resin that the design suggested wasn't working well (we can try the lam we have but if that doesnt work at least we know the epon 862 is a good fallback).

Also he sent this invite to a filament winder discord server made by Andrew Reilly: https://discord.gg/vPHHyqNw

As for fiber spreading the main points the paper makes are that the entry and exit angles for the fibers relative to the "spreading fixtures" aka metal rods need to be >20 deg and the spreading needs to happen >0.5m from the roll
I've attached a screenshot of the relative pages for that (page 39 of the paper itself but 63 of the pdf)

appendix A also had several other options for mechanical spreading, including:

- A vibration based spreader in which the tow zig zags up and down across several pins while being blasted with 32-39hz tones at 80-130dB
- Using convex rollers
- Using combs with non parallel lines that spread the tow out
- Putting multiple convex rollers on a larger wheel and spinning that wheel as well
- Putting ridges axially across the roller pins

An S shaped mechanical spreader in which the fibers go around two aluminum rods should be pretty easily doable, that can be put in the position of the second roller since that already has two holes in the bracket. I'll do more research on rod diameters and can do some tests based on what stock we have as well. May be worth turning the rods on the lathe to make sure we get a nice smooth surface as opposed to the extruded stock surface.